umiddelb/armhf #1: Thank you for proofreading

umiddelb, 2015-02-15:
@ewalbridge
Thank you for proofreading!

ewalbridge, 2015-02-19:
No problem, glad to help.
Your tutorial was very helpful.

There was one command that I had to run that was different on the ODROID-C1 section.

$ git clone https://github.com/hardkernel/linux --single-branch -b odroidc-3.10.y
$ cd linux
$ make odroidc_defconfig

$ make menuconfig

$ make clean
$ make -j 4 uImage modules
~~$ make meson8b_odroidc.dtb~~
**$ make dtbs**
$ sudo make modules_install
$ sudo cp arch/arm/boot/uImage /media/boot
$ sudo cp arch/arm/boot/dts/meson8b_odroidc.dtb /media/boot

Other that that, it worked perfect.
Thank you for taking the time to write the tutorial.

umiddelb, 2015-02-19:
very good, this saves us from performing that strange kind of [limbo dancing](https://github.com/umiddelb/armhf/wiki/Installing,-running,-using-docker-on-armhf-(ARMv7)-devices#what-if-you-get-compilermake-errors) caused by the aufs patchwork.

```
root@odroid-c1:~/kernel/3.10.65/linux# make dtbs
  CC      scripts/mod/devicetable-offsets.s
  GEN     scripts/mod/devicetable-offsets.h
  HOSTCC  scripts/mod/file2alias.o
  HOSTLD  scripts/mod/modpost
  DTC     arch/arm/boot/dts/meson8b_odroidc.dtb
root@odroid-c1:~/kernel/3.10.65/linux# make meson8b_odroidc.dtb
  CC      scripts/mod/devicetable-offsets.s
  GEN     scripts/mod/devicetable-offsets.h
  HOSTCC  scripts/mod/file2alias.o
  HOSTLD  scripts/mod/modpost
scripts/Makefile.build:128: kbuild: Makefile.build is included improperly
  CC      kernel/bounds.s
  GEN     /include/generated/bounds.h
  CC      arch/arm/kernel/asm-offsets.s
  GEN     /include/generated/asm-offsets.h
make[1]: *** No rule to make target `/missing-syscalls', needed by `__build'.  Stop.
make: *** [meson8b_odroidc.dtb] Error 2
root@odroid-c1:~/kernel/3.10.65/linux# 
```

closed 2015-02-22 by umiddelb