umiddelb/armhf #2: Thanks + questions

kennethlimcp, 2015-02-22:
Hey @umiddelb,

great work for the documentation! Really interesting and helps me figure out what went wrong for my image.

I have a question if you can help me along:

1.) I did not compile my own kernel (so no aufs)
2.) Using a Odroid-C1 minimal from http://odroid.com/dokuwiki/doku.php?id=en:c1_ubuntu_minimal

I have a few errors like `CONFIG_CGROUP_CPUACCT: missing` in my attempt to fire up a container with your armhf image.

Did you experience something similar on the C1 or is there something that i have overlooked?

3.) I used the binary for the 1.5.0 docker version you compiled

Thanks!!

umiddelb, 2015-02-22:
The kernel image provided by hardkernel isn't prepared for docker use. As long as the requirements marked as 'Generally Necessary' are not met, docker won't start. In other words, you need to compile your own kernel. The easiest way to do this is to take the kernel sources from my github repository which includes the necessary kernel options as well as aufs support:

```
git clone https://github.com/umiddelb/linux --single-branch -b odroidc-3.10.y
cd linux
make odroidc_defconfig
make -j 4 uImage modules
make dtbs
sudo make modules_install
sudo cp arch/arm/boot/uImage /media/boot
sudo cp arch/arm/boot/dts/meson8b_odroidc.dtb /media/boot
sudo init 6
```
and you're done.

umiddelb, 2015-02-22:
If you really don't want to compile a custom kernel, you may try my minimal image for the c1 [here](http://forum.odroid.com/viewtopic.php?f=111&t=9038#p74143). It has builtin docker support.

kennethlimcp, 2015-02-22:
Thanks for the information!  In the wiki you mentioned that we can do it using the kernel provided by hardkernel.

Is that still possible now that you recommend the  kernel is from your repo?

Also, can you share how those requirements are patched so that I can learn something new?

Thanks!  :)

umiddelb, 2015-02-22:
I have to be more precise, 'kernel provided by hardkernel' means the kernel source files. You have to modify the kernel configuration at least. But you really want to include aufs support, it will speed up container related operations dramatically.

My sources are in sync with upstream (with the exception of aufs support of course), you can see this on github. There are two modifications:

1. I've enabled kernel options needed for docker to start properly. This step doesn't require patching source files, you enable certain kernel configurations (which are disabled by the vendor) only. `check-config.sh` will tell you the missing options (e.g. `CONFIG_CGROUP_CPUACCT`). 
  
2. I've integrated aus support. This requires [patching the kernel sources](https://github.com/umiddelb/armhf/wiki/Installing,-running,-using-docker-on-armhf-(ARMv7)-devices#full-integration-into-the-linux-kernel-sources-310x). 

All of this is already done in my fork of the kernel sources.

kennethlimcp, 2015-02-22:
Got it! I probably should take a look at how you enabled those configuration.

It will be easy now that you have done most of the heavylifting!

It's on my to learn list on for compiling a distro from source so i guess the has come for me to do it!

Thanks for your patience though with a beginner :)

umiddelb, 2015-02-23:
This is only about how to compile a custom kernel which is expected to happen on the ARM platform.
On the i386 or amd64 platform the mainstream distributions have included almost everything in their kernel images which makes compiling your own kernel very unlikely. On ARM storage requirements matter (especially when you want to put your kernel image in NAND flash memory), so the vendors publish kernel configurations which lead to smaller images.

If you want to compile the whole distribution (not only the kernel) then you should become familiar with gentoo or 'linux from scratch' (or even freebsd). ubuntu, archlinux, debian or fedora use precompiled packages.

kennethlimcp, 2015-02-23:
@umiddelb, just took a look at your forked kernel repo and it is even with the upstream.

I don't see where are the changes you did to enable the cgroups requirement. 

Or they are enabled by default already on the upstream kernel?

umiddelb, 2015-02-23:
The kernel configuration is stored in a file named `.config`
The command `make odroidc_defconfig` will generate a new version of the file derived from a template file. You can find this template in `./arch/arm/configs/odroidc_defconfig`. You can find the modifications I made to the kernel config there.

https://github.com/umiddelb/linux/commit/49df9ea7f36303c4221d04e9d581938b8d876b0c#diff-9a711c0d0e135f2c8bd55323133b09c1

kennethlimcp, 2015-02-24:
@umiddelb, i saw the changes to `odroid_defconfig` in the commit but looking at the latest branch `odroidc3.10.y` which you recommended for kernel compilation, the settings are not there and defaults to hardkernel version. 

Am i missing something? :dancers: 

https://github.com/umiddelb/linux/blob/odroidc-3.10.y/arch/arm/configs/odroidc_defconfig

umiddelb, 2015-02-24:
@kennethlimcp 

please try it again, it should work now:
```
git clone https://github.com/umiddelb/linux --single-branch -b odroidc-3.10.y
cd linux
make odroidc_defconfig
make -j 4 uImage modules
make dtbs
sudo make modules_install
sudo cp arch/arm/boot/uImage /media/boot
sudo cp arch/arm/boot/dts/meson8b_odroidc.dtb /media/boot
sudo init 6
```

You can see the differences [here](https://github.com/hardkernel/linux/pull/84/files), just scroll down until odroidc_defconfig begins (search for arch/arm/configs/odroidc_defconfig in your browser) ...

kennethlimcp, 2015-02-24:
@umiddelb, cool! guess when you merged upstream it got reverted back to the default file that's why i didn't see it. Hope it ends up in the hardkernel repo :)

Thanks for the help! I'll try to compile them tonight and update you again

umiddelb, 2015-02-24:
@kennethlimcp 

It's [merged](https://github.com/hardkernel/linux/pull/84) into the hardkernel repo now.

kennethlimcp, 2015-02-26:
@umiddelb, just wanted to build and realized your linux repo is gone! Are you recommended the Hardkernel repo instrad? :)

umiddelb, 2015-02-26:
@kennethlimcp

yes of course, It's the official source for the c1 linux kernel. There is no reason to have a fork since the pull request ist processed successfully.

kennethlimcp, 2015-02-26:
Some errors with `make odroidc_defconfig` though on Mac osx: 

```
10 warnings generated.
  HOSTCC  scripts/kconfig/zconf.tab.o
  HOSTLD  scripts/kconfig/conf
***
*** Can't find default configuration "arch/x86/configs/odroidc_defconfig"!
***
make[1]: *** [odroidc_defconfig] Error 1
make: *** [odroidc_defconfig] Error 2
```

umiddelb, 2015-02-26:
It's easier when you compile the kernel on the c1 directly ... otherwise you need to have everything in place for cross compiling

kennethlimcp, 2015-02-26:
Just ran it on the C1 but not sure why `sudo cp arch/arm/boot/uImage /media/boot` is not working. the uImage folder wasn't generated.

Re-running a clean build again ;)

umiddelb, 2015-02-26:
you may check, if you have the tool

bc

installed. It's needed for building the kernel image and not part of the default installation.

kennethlimcp, 2015-02-26:
Good point. i ran the `make uImage` and it showed me the error so installed it.

there's another issue with using files from `.zip` since i wanted to save time downloading but seems like i have to use `git clone` for everything to work well.

Sorry for the trouble! Hope this builds fine!

umiddelb, 2015-02-26:
I'm not sure if zip will archive files/directories starting with a '.' by default. You migth have missed important files (e.g. .config).

zip is not very common, most people use tar

kennethlimcp, 2015-02-27:
@umiddelb,

1.) i managed to compile the kernel on C1. Missing packages are `bc` and `lzop` for uImage to compile
2.) Using the native `apt-get install docker.io` i managed to run the hello world successfully
3) However, after performing a docker 1.5.0 update, building a Dockerfile gave me `INFO[0000] mountpoint for cpuset not found`

That sounds like a kernel issue?

umiddelb, 2015-02-27:
please make sure, that you have install the following packages (not only docker.io):
`sudo apt-get install lxc aufs-tools cgroup-lite apparmor docker.io`

and then reboot

kennethlimcp, 2015-02-27:
@umiddelb, i just ran the commands and they are indeed installed already. Also tried to use the `docker-1.5.0` binary you compiled but resulted in the same error.

I looked at the PR you sent and seems like it was indeed not set:

https://github.com/hardkernel/linux/blob/odroidc-3.10.y/arch/arm/configs/odroidc_defconfig#L109

umiddelb, 2015-02-27:
could you please send the output of 
check-config.sh
and the whole docker error message (from docker -d -D)?

kennethlimcp, 2015-02-28:
ok i will do it when i compile the kernel again. Thought everything was working and deleted the folder!

kennethlimcp, 2015-02-28:
Checking the **.config** file after the command `make odroidc_defconfig`

```bash
info: reading kernel config from .config ...

Generally Necessary:
- cgroup hierarchy: properly mounted [/sys/fs/cgroup]
- CONFIG_NAMESPACES: enabled
- CONFIG_NET_NS: enabled
- CONFIG_PID_NS: enabled
- CONFIG_IPC_NS: enabled
- CONFIG_UTS_NS: enabled
- CONFIG_DEVPTS_MULTIPLE_INSTANCES: enabled
- CONFIG_CGROUPS: enabled
- CONFIG_CGROUP_CPUACCT: enabled
- CONFIG_CGROUP_DEVICE: enabled
- CONFIG_CGROUP_FREEZER: enabled
- CONFIG_CGROUP_SCHED: enabled
- CONFIG_MACVLAN: enabled
- CONFIG_VETH: enabled
- CONFIG_BRIDGE: enabled
- CONFIG_NF_NAT_IPV4: enabled
- CONFIG_IP_NF_FILTER: enabled
- CONFIG_IP_NF_TARGET_MASQUERADE: enabled
- CONFIG_NETFILTER_XT_MATCH_ADDRTYPE: enabled
- CONFIG_NETFILTER_XT_MATCH_CONNTRACK: enabled
- CONFIG_NF_NAT: enabled
- CONFIG_NF_NAT_NEEDED: enabled
- CONFIG_POSIX_MQUEUE: enabled

Optional Features:
- CONFIG_MEMCG_SWAP: enabled
- CONFIG_RESOURCE_COUNTERS: enabled
- CONFIG_CGROUP_PERF: enabled
- Storage Drivers:
  - "aufs":
    - CONFIG_AUFS_FS: enabled
    - CONFIG_EXT4_FS_POSIX_ACL: enabled
    - CONFIG_EXT4_FS_SECURITY: enabled
  - "btrfs":
    - CONFIG_BTRFS_FS: enabled
  - "devicemapper":
    - CONFIG_BLK_DEV_DM: enabled
    - CONFIG_DM_THIN_PROVISIONING: enabled
    - CONFIG_EXT4_FS: enabled
    - CONFIG_EXT4_FS_POSIX_ACL: enabled
    - CONFIG_EXT4_FS_SECURITY: enabled
  - "overlay":
    - CONFIG_OVERLAY_FS: missing
    - CONFIG_EXT4_FS_SECURITY: enabled
    - CONFIG_EXT4_FS_POSIX_ACL: enabled
```

umiddelb, 2015-02-28:
OK, looks good as it is. Now it's time to tidy up a little bit. I assume you don't have any valuable docker images you want to keep (otherwise you open a duckerhub account and push your images there). Having that said, you run the following commands:

    sudo service docker.io stop
    sudo rm -rf /var/lib/docker
    sudo service docker.io start

kennethlimcp, 2015-02-28:
just did and the same error persists. :) 

Instead of wasting your time, maybe i do everything from scratch again and hopefully reproduce the exact same issue!

```
INFO[0000] +job serveapi(unix:///var/run/docker.sock)   
INFO[0000] Listening for HTTP on unix (/var/run/docker.sock) 
DEBU[0000] Registering GET, /info                       
DEBU[0000] Registering GET, /version                    
DEBU[0000] Registering GET, /images/viz                 
DEBU[0000] Registering GET, /containers/{name:.*}/changes 
DEBU[0000] Registering GET, /containers/{name:.*}/stats 
DEBU[0000] Registering GET, /exec/{id:.*}/json          
DEBU[0000] Registering GET, /containers/{name:.*}/json  
DEBU[0000] Registering GET, /containers/{name:.*}/attach/ws 
DEBU[0000] Registering GET, /_ping                      
DEBU[0000] Registering GET, /events                     
DEBU[0000] Registering GET, /images/json                
DEBU[0000] Registering GET, /images/{name:.*}/json      
DEBU[0000] Registering GET, /containers/json            
DEBU[0000] Registering GET, /containers/{name:.*}/export 
DEBU[0000] Registering GET, /containers/ps              
DEBU[0000] Registering GET, /containers/{name:.*}/top   
DEBU[0000] Registering GET, /containers/{name:.*}/logs  
DEBU[0000] Registering GET, /images/search              
DEBU[0000] Registering GET, /images/get                 
DEBU[0000] Registering GET, /images/{name:.*}/get       
DEBU[0000] Registering GET, /images/{name:.*}/history   
DEBU[0000] Registering POST, /containers/{name:.*}/unpause 
DEBU[0000] Using graph driver aufs                      
DEBU[0000] Migrating existing containers                
DEBU[0000] Creating images graph                        
DEBU[0000] Restored 0 elements                          
DEBU[0000] Creating repository list                     
INFO[0000] +job init_networkdriver()                    
DEBU[0000] /sbin/iptables, [--wait -C POSTROUTING -t nat -s 172.17.42.1/16 ! -o docker0 -j MASQUERADE] 
DEBU[0000] Registering POST, /containers/{name:.*}/copy 
DEBU[0000] Registering POST, /exec/{name:.*}/resize     
DEBU[0000] Registering POST, /build                     
DEBU[0000] Registering POST, /containers/create         
DEBU[0000] Registering POST, /containers/{name:.*}/start 
DEBU[0000] Registering POST, /containers/{name:.*}/stop 
DEBU[0000] Registering POST, /containers/{name:.*}/rename 
DEBU[0000] Registering POST, /images/create             
DEBU[0000] Registering POST, /containers/{name:.*}/restart 
DEBU[0000] Registering POST, /containers/{name:.*}/wait 
DEBU[0000] Registering POST, /containers/{name:.*}/attach 
DEBU[0000] Registering POST, /containers/{name:.*}/exec 
DEBU[0000] Registering POST, /exec/{name:.*}/start      
DEBU[0000] Registering POST, /auth                      
DEBU[0000] Registering POST, /images/load               
DEBU[0000] Registering POST, /images/{name:.*}/push     
DEBU[0000] Registering POST, /images/{name:.*}/tag      
DEBU[0000] Registering POST, /containers/{name:.*}/kill 
DEBU[0000] /sbin/iptables, [--wait -D FORWARD -i docker0 -o docker0 -j DROP] 
DEBU[0000] Registering POST, /containers/{name:.*}/pause 
DEBU[0000] Registering POST, /containers/{name:.*}/resize 
DEBU[0000] Registering POST, /commit                    
DEBU[0000] Registering DELETE, /containers/{name:.*}    
DEBU[0000] Registering DELETE, /images/{name:.*}        
DEBU[0000] Registering OPTIONS,                         
DEBU[0000] docker group found. gid: 107                 
DEBU[0000] /sbin/iptables, [--wait -C FORWARD -i docker0 -o docker0 -j ACCEPT] 
DEBU[0000] /sbin/iptables, [--wait -C FORWARD -i docker0 ! -o docker0 -j ACCEPT] 
DEBU[0000] /sbin/iptables, [--wait -C FORWARD -o docker0 -m conntrack --ctstate RELATED,ESTABLISHED -j ACCEPT] 
DEBU[0000] /sbin/iptables, [--wait -t nat -D PREROUTING -m addrtype --dst-type LOCAL -j DOCKER] 
DEBU[0000] /sbin/iptables, [--wait -t nat -D OUTPUT -m addrtype --dst-type LOCAL ! --dst 127.0.0.0/8 -j DOCKER] 
DEBU[0000] /sbin/iptables, [--wait -t nat -D OUTPUT -m addrtype --dst-type LOCAL -j DOCKER] 
DEBU[0000] /sbin/iptables, [--wait -t nat -D PREROUTING -j DOCKER] 
DEBU[0000] /sbin/iptables, [--wait -t nat -D OUTPUT -j DOCKER] 
DEBU[0000] /sbin/iptables, [--wait -t nat -F DOCKER]    
DEBU[0000] /sbin/iptables, [--wait -t nat -X DOCKER]    
DEBU[0000] /sbin/iptables, [--wait -t nat -n -L DOCKER] 
DEBU[0000] /sbin/iptables, [--wait -t nat -N DOCKER]    
DEBU[0000] /sbin/iptables, [--wait -C -m addrtype --dst-type LOCAL] 
DEBU[0000] /sbin/iptables, [--wait -t nat -A PREROUTING -m addrtype --dst-type LOCAL -j DOCKER] 
DEBU[0000] /sbin/iptables, [--wait -C -m addrtype --dst-type LOCAL ! --dst 127.0.0.0/8] 
DEBU[0000] /sbin/iptables, [--wait -t nat -A OUTPUT -m addrtype --dst-type LOCAL ! --dst 127.0.0.0/8 -j DOCKER] 
DEBU[0000] /sbin/iptables, [--wait -t filter -n -L DOCKER] 
DEBU[0000] /sbin/iptables, [--wait -C FORWARD -o docker0 -j DOCKER] 
INFO[0000] -job init_networkdriver() = OK (0)           
DEBU[0000] Restarting containers...                     
INFO[0000] docker daemon: 1.5.0 a8a31ef-dirty; execdriver: native-0.2; graphdriver: aufs 
INFO[0000] +job acceptconnections()                     
INFO[0000] -job acceptconnections() = OK (0)    
```

umiddelb, 2015-02-28:
But this is looking good now ...

kennethlimcp, 2015-02-28:
That's the funny thing! 1.0.1 works well but not 1.5.0.

on a sidenote...this is my motivation for learning all these armv7 stuff: https://github.com/kennethlimcp/spark-server-automate

umiddelb, 2015-02-28:
What is the exact error message you get?

If you want to set up your system from scratch, you may try this [approach](https://github.com/umiddelb/armhf/wiki/Install-Ubuntu-Core-14.04-on-ARMv7-(ODROID-C1)), but you need to compile the kernel from source once again afterwards.

closed 2015-03-12 by umiddelb