umiddelb/armhf #4: .img for cubox-i

kennethlimcp, 2015-04-10:
Hey @umiddelb, i downloaded the image from your website for Cubox-i (actually running on hummingboard) but the `.bz2` file gave me a `.tar` which uncompressed to the files for the SD.

Is that the right way to install by copying the files over to the SD? How do we expand and use the entire diskspace then? Thanks!

umiddelb, 2015-04-11:
Hi @kennethlimcp,

the tar archive contains the contents of the root filesystem. So you create the filesystem by yourself and extract the tar archive on the newly created filesystem.
```
sudo mkfs.ext4 -O ^has_journal -b 4096 -L rootfs /dev/sdb1
sudo mount /dev/sdb1 /mnt
curl -sSL https://www.dropbox.com/sh/rfxvv8ayozeitlu/AABi5TPly63oqg-PLXO8p_4Ha/trusty.tar.bz2?dl=0 | sudo tar --numeric-owner -xpjf - -C /mnt
```
Please remember to install the u-boot bootloader as well, if you haven't used the uSD with the CuBox-i before, see [here](http://wiki.solid-run.com/Building_the_kernel_and_u-boot_for_the_CuBox-i_and_the_HummingBoard).

kennethlimcp, 2015-04-11:
Ok thanks!

closed 2015-04-11 by kennethlimcp

umiddelb, 2015-04-12:
What kind of Hummingboard do you have? The tar archive contains two dtb files for the Hummingboard, but I'm not sure if the bootloader for the CuBox-i does also fit the Hummingboard. Can you see some bootloader messages when the Hummingboard start up?

kennethlimcp, 2015-04-12:
I have a i1 and i2eX. Once i get it on the sd card i will test it out and let you know ;)