umiddelb/armhf #5: [C1] cgroup hierarchy: nonexistent??

kennethlimcp, 2015-04-12:
Hey @umiddelb,

i also tried out the latest Ubuntu from Hardkernel and i ran the Docker check just to be sure.

Seems like by default, they did not enable the `- cgroup hierarchy: nonexistent??`

There's a script to do so as mentioned:  https://github.com/tianon/cgroupfs-mount

So i thought i would mention to you in case someone asked the same thing. I will test out Docker 1.5.0 and see how it goes and report back.

Thanks!

umiddelb, 2015-04-12:
Hi @kennethlimcp 

At least you have to make sure that the following packages are installed:
```
sudo apt-get install lxc aufs-tools cgroup-lite apparmor docker.io 
```
cgroup hierarchy is provided by the package cgroup-lite not by the kernel.

kennethlimcp, 2015-04-12:
That worked thanks! :)

closed 2015-04-12 by kennethlimcp

umiddelb, 2015-04-12:
you're welcome! Which Ubuntu Image did you take?

kennethlimcp, 2015-04-12:
I'm using the 14.04 image from Hardkernel. Managed to update to Docker 1.5.0 as well :)

umiddelb, 2015-04-12:
so you've taken the image from [here](http://odroid.in/ubuntu_14.04lts/)?

kennethlimcp, 2015-04-12:
Yup that's right!