umiddelb/armhf #8: Cubox-i Pro make process is broken

aerodame, 2015-08-20:
$ make imx_v7_cbi_hb_defconfig does not succeed... -->
***
*** Can't find default configuration "arch/x86/configs/imx_v7_cbi_hb_defconfig"!
***

Any idea where to go with this make configuration update?

umiddelb, 2015-08-20:
> $ make imx_v7_cbi_hb_defconfig does not succeed... -->
> 
> *** Can't find default configuration "arch/x86/configs/imx_v7_cbi_hb_defconfig"!
> 
> Any idea where to go with this make configuration update?

This happens when you try to compile an ARM target on x86 without having set up cross compiling.

Please try to compile the kernel on the CuBox-i directly, it will take about 1h 30minutes.

aerodame, 2015-08-21:
Ah...very well.  So I moved on to the CuBox and got through to this point with the following error message:

$ make -j 4 zImage
...
INSTALL include/linux/can (5 files)
  INSTALL include/scsi/fc (4 files)
  INSTALL include/linux/dvb (8 files)
  INSTALL include/scsi (3 files)
  INSTALL include/linux/hdlc (1 file)
  INSTALL include/sound (11 files)
  INSTALL include/video (3 files)
  INSTALL include/linux/hsi (1 file)
  INSTALL include/xen (4 files)
  INSTALL include/linux/isdn (1 file)
  INSTALL include/linux/mmc (1 file)
  INSTALL include/uapi (0 file)
/root/linux-imx6-3.14/scripts/Makefile.headersinst:55: *** Missing UAPI file /root/linux-imx6-3.14/include/uapi/linux/netfilter/xt_CONNMARK.h.  Stop.
/root/linux-imx6-3.14/scripts/Makefile.headersinst:128: recipe for target 'netfilter' failed
make[3]: *** [netfilter] Error 2
/root/linux-imx6-3.14/scripts/Makefile.headersinst:128: recipe for target 'linux' failed
make[2]: *** [linux] Error 2
make[2]: *** Waiting for unfinished jobs....
/root/linux-imx6-3.14/Makefile:964: recipe for target 'headers_install' failed
make[1]: *** [headers_install] Error 2
Makefile:819: recipe for target 'vmlinux' failed
make: *** [vmlinux] Error 2
root@cubox-i:~/linux-imx6-3.14# uname -a
Linux cubox-i 3.14.14-cubox-i #8 SMP Sun Jul 12 19:03:09 CEST 2015 armv7l GNU/Linux

Any ideas about this ?

umiddelb, 2015-08-21:
Please check, if you have installed `bc` and `lzop`.

aerodame, 2015-08-22:
OK, installed lzop (already had bc).   The compiled seemed to finish OK.  Can you clarify what the next steps are?  Is there an image created somewhere that I need to reflash as the new OS or is this build process supposedly modifying the existing OS in place?  I rebooted, but there is no docker present.

umiddelb, 2015-08-22:
The steps mentioned [here](https://github.com/umiddelb/armhf/wiki/Installing,-running,-using-docker-on-armhf-(ARMv7)-devices#cubox-i-pro) will help you to compile and install a custom linux kernel on the CuBox-i. The step
```
$ sudo cp arch/arm/boot/zImage arch/arm/boot/dts/*.dtb /boot
```
copies the kernel image and the device tree binaries to the directory `/boot`. 

Most of the ARM devices use [u-boot](http://www.denx.de/wiki/U-Boot) as the boot manager. The default configuration on the CuBox-i will search for a file called `zImage` on the first partition of the µSD (either in `/`or in `/boot`). I haven't used the linux images provided by SolidRun, so I don't know their exact partitioning layout. Usually the first partition is mounted as `/boot`, the second is mounted as `/`.

The above relates to the linux kernel only, the rest of the operating system is not affected.

If you want to run Docker you need to install the relevant packages
```
$ sudo apt-get install lxc aufs-tools cgroup-lite apparmor docker.io
```

To complete the setup, you may [integrate aufs support into your custom kernel](https://github.com/umiddelb/armhf/wiki/Installing,-running,-using-docker-on-armhf-(ARMv7)-devices#aufs-integration) and [update the docker binary](https://github.com/umiddelb/armhf/wiki/Installing,-running,-using-docker-on-armhf-(ARMv7)-devices#updating-from-docker-version-101).

aerodame, 2015-08-22:
Thanks for the continued help on this... one of the things that seems to be still missing is any package support from ARM for docker.io...   Did you get this message at some point?

root@cubox-i:~# sudo apt-get install lxc aufs-tools cgroup-lite apparmor docker.io
Reading package lists... Done
Building dependency tree       
Reading state information... Done
Package docker.io is not available, but is referred to by another package.
This may mean that the package is missing, has been obsoleted, or
is only available from another source

E: Unable to locate package cgroup-lite
E: Package 'docker.io' has no installation candidate

umiddelb, 2015-08-22:
It seems you're using debian. The tutorial is written for ubuntu users.

umiddelb, 2015-08-22:
for Debian, you may try step 4 mentioned [here](https://github.com/umiddelb/armhf/wiki/Get-Docker-up-and-running-on-the-RaspberryPi-%28ARMv6%29-in-four-steps).

umiddelb, 2015-08-25:
If you want to try ubuntu, you may take the image I'm using for my CuBox-i. You can find it [here](http://cubox.middelberg.de).

closed 2015-08-25 by umiddelb