umiddelb/armhf #9: Trying docker 1.8.1 Fails

aerodame, 2015-08-25:
Uli, still compiling 1.7.1 but initially I tried building 1.8.1 and received the following error...

$ git clone -b 'v1.8.1' --single-branch https://github.com/docker/docker.git 
$ sudo make build

... 
Step 14 : RUN curl -sSL ...
2015/08/25 16:17:23 **finalize namespace drop capabilities operation not permitted** 
2015/08/25 18:17:23 The command [/bin/sh -c curl -sSL https://golang.org/dl/go${GO_VERSION}.src.tar.gz | tar -v -C /usr/local -xz && mkdir -p /go/bin] **returned a non-zero code: 1**
make: *** [build] Error 1

Just curious if you have seen the same and/or know how to get past this error.

umiddelb, 2015-08-25:
Steve, the error you're getting (finalize namespace ...) is a known bug in Docker 1.0.1.
Just repeat 'make build', the error doesn't occur every time you invoke docker. You can download my Docker 1.7.1 binary for ARM available on GitHub, if you're impatient.

Docker 1.8.0/1 doesn't compile on ARM for various reasons, there are some open issues on GitHub.

aerodame, 2015-08-25:
I did finally download your binary and am starting to test it now...  Exhausting process.  Thanks!

umiddelb, 2015-08-25:
>I did finally download your binary and am starting to test it now...  Exhausting process.  Thanks!

I've tried to automate the process setting up ubuntu core + docker for the cubox-i (and other ARMv7 devices as well). You may take a look [here](/umiddelb/z2d). The [xilka kernel](http://xilka.com/kernel) works well with docker.

umiddelb, 2015-09-08:
I believe you're fine running Docker on your CuBox.
I'll close this issue now.

closed 2015-09-08 by umiddelb