umiddelb/armhf #10: About ARMv6 binany static-link file

philipz, 2015-09-09:
Hi @umiddelb , thanks for your docker binary file. It help me to build ARMv7 Raspberry Pi 2 version of weave, https://github.com/philipz/weave . But when build Raspberry Pi ARMv6 Arch. need ARMv6 docker binary  static-link file, like https://github.com/umiddelb/armhf/blob/master/bin/arm7/docker-1.5.0 . Would you provide ARMv6 binany file? Or provide build steps document. Thank you!

umiddelb, 2015-09-09:
Hi @philipz, you're welcome. The hypriot team provides docker install packages for ARMv6 (see [here](https://github.com/umiddelb/armhf/wiki/Get-Docker-up-and-running-on-the-RaspberryPi-%28ARMv6%29-in-four-steps) for more details), but the Docker binary there is linked dynamically.
If you need to have a static-linked binary, you need to tweak their [build process](https://github.com/hypriot/rpi-docker-builder). Docker uses ubuntu as the default build environment (which isn't available for the RPi 1).

Cheers,
Uli

philipz, 2015-09-10:
The [rpi-docker-builder](https://github.com/hypriot/rpi-docker-builder) is awesome. Thank you!!

closed 2015-09-10 by philipz