umiddelb/armhf #12: Cannot proceed with Install Fedora 21 on ARMv7 (ODROID C1)

jdeisenberg, 2015-09-18:
All the steps work until this one:

    ( cd /mnt; sudo find * | sudo cpio -dump .../rootfs )

I get many messages of the form:

    cpio: cannot make directory `...': Read-only file system

I'm also curious why you have three dots in a row instead of two.

umiddelb, 2015-09-18:
`.../rootfs` means an arbitrary path pointing to a directory called `rootfs` somewhere else in your filesystem (which you have created before starting `cpio`).

Fedora 21 has become outdated now. Have you tried this [fedora 22](http://forum.odroid.com/viewtopic.php?f=114&t=8872) image?

jdeisenberg, 2015-09-20:
Your link leads to a Fedora 21 image.

umiddelb, 2015-09-20:
Sorry, I haven't seen that. Then you should start with the fedora 21 image for the C1.

closed 2015-09-23 by umiddelb