umiddelb/armhf #13: Please build docker-1.9.0 for arm.

vimagick, 2015-11-04:
I'm trying to build it on my own.

```
$ ./build --arm 6 1.9.0
...
Fetched 2527 kB in 1min 3s (40.1 kB/s)
Reading package lists...
Reading package lists...
Building dependency tree...
Reading state information...
E: Unable to locate package ubuntu-zfs
The command '/bin/sh -c apt-get update && apt-get install -y    apparmor        aufs-tools      automake        bash-completion    btrfs-tools     build-essential         createrepo      curl    dpkg-sig        git     iptables        libapparmor-dev    libcap-dev      libsqlite3-dev  mercurial       parallel        python-mock     python-pip      python-websocket  reprepro         ruby1.9.1       ruby1.9.1-dev   s3cmd=1.1.0*    ubuntu-zfs      libzfs-dev      --no-install-recommends' returned a non-zero code: 100
Makefile:74: recipe for target 'build' failed
make: *** [build] Error 1
```

I found there is no `ubuntu-zfs_8~trusty_armhf.deb` in [PPA][1].
I don't know how to solve it. Please help. Thanks!

[1]: http://ppa.launchpad.net/zfs-native/stable/ubuntu/pool/main/u/ubuntu-zfs/

vimagick, 2015-11-04:
I didn't notice that [Dockerfile][1] was modified further. I'll try it again.

[1]: https://github.com/umiddelb/armhf/blob/master/Dockerfile.armv7#L59-L60

umiddelb, 2015-11-04:
I'm about to build Docker 1.9.0 for ARMv7, just wait a couple of minutes ...

vimagick, 2015-11-04:
Thank you very much! :+1:

umiddelb, 2015-11-04:
done

closed 2015-11-04 by umiddelb

vimagick, 2016-01-15:
:smile: I've built [docker-1.9.1](https://github.com/vimagick/rpi-bin/wiki/docker) successfully.

umiddelb, 2016-01-16:
starting with docker version 1.10 docker for arm will be build by the [official repository](https://github.com/docker/docker). You can try it with the current development tree.

aerodame, 2016-01-16:
Uli we built up a 1.9.1 on a bunch of CuBox units and have this consistent devmapper fault that occurs all the time when building images from Dockerfiles.   Have you seen that issue?   You can succeed in eventually completing images by repeating the docker build until it complete but it is a royal pain.

Not sure which file system would be best for these mx6 arm devices.

umiddelb, 2016-01-17:
Hi Steve,
hm, this sounds weird. Do you mind opening a new github issue and sharing some information about your environment:
- which device: CuBox (Marvell Armada5XX based) or CuBox-i (Freescale/NXP imx6 based)?
- which kernel (source code repository)?
- the output of [check-config.sh](https://raw.githubusercontent.com/docker/docker/master/contrib/check-config.sh)

Docker can use several storage backends, the devicemapper is used as a last resort (it's really slow).
I'm using overlayfs on my CuBox-i running a 4.4 vanilla mainline Linux kernel. I haven't used Docker 1.9.1 so far, only 1.9.0 and 1.10.0-dev.

aerodame, 2016-01-17:
Will do!