umiddelb/armhf #15: Incorrect statement regarding odroid xu4 kernel

rickyzhang82, 2015-12-24:
The statement (https://github.com/umiddelb/armhf/wiki/Installing,-running,-using-docker-on-armhf-(ARMv7)-devices) that odroid stock kernel support AUFS is not correct any more. I upgrade my u2 to xu4 with stock kernel. When reusing root file system with docker container there, I found that default kernel config in xu4 doesn't support AUFS any more. Please update your statement.

rickyzhang82, 2015-12-24:
In addition, the default odroid xu3 config doesn't enable several cgorup options, either.

umiddelb, 2015-12-25:
Hi @rickyzhang82,

this happens from time to time, but it is going to be fixed by the kernel maintainer. I have reopened the corresponding [issue](https://github.com/hardkernel/linux/issues/147). Please open a new issue there if this will happen again.

Merry Christmas!

closed 2016-02-13 by umiddelb