umiddelb/armhf #18: u-boot instructions are not ones that should be used

ausil, 2016-01-30:
Please read http://git.denx.de/?p=u-boot.git;a=blob;f=doc/README.distro;h=3715c8c3bad7339c41a9da7e1f11fd6d4f902a08;hb=HEAD there is a defined way to setup u-boot to ensure that distros can simply support hardware. The instructions you have are not at all close to meeting the specification.

umiddelb, 2016-01-31:
thank you for pointing me to this documentation.

Honestly I wasn't aware that u-boot offers these kinds of facilities (e.g. a boot menu), because most of the implementations I (have to) use have skipped this options. I've started my own solution to harmonize different u-boot usages/implementations across the devices I have. You may take a look [here](https://github.com/umiddelb/u-571). And I will start to re-factor the variable naming in order to be more compliant.

ausil, 2016-03-08:
Happy to send you a patch that converts you to using distro boot defaults. it will mean fedora and Debian at the least will just work

umiddelb, 2016-03-08:
OK, I'd be happy to have a look at their design documents as well (giving some reasoning), if there are any. IMHO each SoC/Board Vendor invents its own way how to utilize u-boot, which is very annoying.

closed 2016-05-23 by umiddelb