umiddelb/armhf #19: Tutorial: "WARNING! I don't seem to be running in the Docker container."

saljut7, 2016-02-28:
Thank you very much for the nice tutorial https://github.com/umiddelb/armhf/wiki/Get-Docker-up-and-running-on-the-RaspberryPi-%28ARMv6%29-in-three-steps but there is one error for `sudo sh run-builder.sh`:

```
HEAD is now at c3959b1... Merge pull request #20526 from tiborvass/1.10.2-cherrypicks
# WARNING! I don't seem to be running in the Docker container.
# The result of this command might be an incorrect build, and will not be
#   officially supported.
#
# Try this instead: make all
# 
```

So in the end: How does it work when you want to build docker for ARM for the first time but you need a docker container for this to do this? So for the beginning I have to use a pre compiled docker package?

umiddelb, 2016-02-28:
> So for the beginning I have to use a pre compiled docker package?
Hm, you may try this then. At the time of writing the build process didn't require Docker to be installed ...

saljut7, 2016-02-28:
Thx for your immediate reply. After purging every docker related stuff I tried again compiling it and the deb file worked fine even there was the warning I've spoken about. So everything is fine now.

But in generel docker on ARM can only run docker images created for ARM, right?

umiddelb, 2016-02-29:
> But in generel docker on ARM can only run docker images created for ARM, right?

correct, it's the same with x86. You can of course use qemu to emulate ARM on x86.
ARMv8(PINE64, ODROID-C2,Raspberry Pi 3)  can execute ARMv7 (most 32 ARM CPUs) and ARMv6 (Raspberry Pi1)

saljut7, 2016-02-29:
Thanks for confirming!
Well, self compiling docker is a good experience for a beginner like me. But is there a community repo I could use for apt-get upgrade docker automatically in the future?

umiddelb, 2016-03-01:
Most Linux distributions already have an installer package for Docker, which includes additional configuration items as well, so `sudo apt-get install docker` should work in most cases.

Latest Docker binaries can be found [here](https://master.dockerproject.org/) as well.

saljut7, 2016-03-03:
Thanks for the Link. 
But just to clarfiy: I was asking for an ARM repository. The fact that I didn't find a docker package in my Debian distribution for ARM led me to your tutorial "docker on a raspberry pi in three steps". ;)

umiddelb, 2016-03-04:
You may need to enable stable / testing repository in order to install Docker on Debian Jessie (stock Debian, not Raspbian) out of the box, please take a look [here](https://github.com/umiddelb/z2d).

saljut7, 2016-03-05:
That's clear. I was just wondering if there are any (community-)repos for Docker on ARM devices. If there are no such repos it's okay - I just wanted to ask someone like you with a lot of experience in Docker.

umiddelb, 2016-03-05:
The Hypriot team maintains a community repo for Docker and Docker related tools (ARMv7/armhf at this time only, ARMv8/aarch64 still to come).

```
wget -q https://packagecloud.io/gpg.key -O - | apt-key add -
echo 'deb https://packagecloud.io/Hypriot/Schatzkiste/debian/ wheezy main' > /etc/apt/sources.list.d/hypriot.list
```

saljut7, 2016-03-06:
Sorry, I could have find this by myself before ( http://blog.hypriot.com/post/family_arm_hardware_for_docker_more_children/ ). Thanks alot, this is what I was looking for!