umiddelb/armhf #22: 'Warning: Bad CRC, using default environment' ?

g40, 2016-04-29:
Thanks for the wiki postings. Very helpful. I'm running a UX4. Any thoughts on the bad CRC message?

```
odroid@odroid:~$ uname -a
Linux odroid 3.10.92-76 #1 SMP PREEMPT Tue Jan 12 16:55:48 BRST 2016 armv7l armv7l armv7l GNU/Linux
```
 
```
odroid@odroid:~$ cat /etc/fw_env.config
# <device>   <offset> <length>
/dev/mmcblk0 0x99E00   0x4000
odroid@odroid:~$
```

```
odroid@odroid:~$ fw_printenv
Warning: Bad CRC, using default environment
bootargs=
bootcmd=bootp; setenv bootargs root=/dev/nfs nfsroot=${serverip}:${rootpath} ip=${ipaddr}:${serverip}:${gatewayip}:${netmask}:${hostname}::off; bootm
bootdelay=3
baudrate=115200
stdin=serial,cros-ec-keyb
stdout=serial,lcd
stderr=serial,lcd
```

umiddelb, 2016-04-29:
This warning is displayed when you try to access an uninitialized on-disk u-boot environment.
If you have serial access you may just interrupt the boot process by typing CR and then:

```
saveenv
reset
``` 

`saveenv` stores the current u-boot working environment on disk. During startup u-boot initializes the working environment with the environment stored on disk, if this fails u-boot falls back to the built-in environment. IMHO the HK images are shipped with an uninitialized (or empty) u-boot environment.

umiddelb, 2016-04-29:
> Linux odroid 3.10.92-76

Btw. your kernel package isn't up to date.

g40, 2016-04-29:
Hello Uli and thanks again. saveenv/reset works perfectly.

>  your kernel package isn't up to date.

That's odd. https://github.com/mdrjr/odroid-utility claims it is. Which version are you running and how did you install it?

Thanks again.

umiddelb, 2016-04-29:
Kernel version 3.10.96 is available.

You should have installed the kernel via

    sudo apt-get install linux-image-xu3

It might be the case that the install package is held back, then

    sudo apt-get upgrade linux-image-xu3
 
or manual interaction with `sudo aptitude` should fix this.

g40, 2016-04-29:
Thanks again Uli. I'll not hijack this issue any longer as it is definitely closed. However I tried the `install linux-image-xu3` and it again claimed the kernel was up to date. I'll worry about that later :)

closed 2016-04-29 by g40