umiddelb/armhf #24: rt patch for Odroid XU4 mainline

FabianBeck, 2016-06-09:
Hey @umiddelb,

Thank you for the great Wikipage :-)
I am a kernel compilation newbie and I would like to compile the mainline kernel with a realtime patch for the Odroid XU4. I found out that the newest Kernelversion.Patchlevel.Subslevel is  4.6.0.
I searched for the corresponding rt-patch and found 
https://www.kernel.org/pub/linux/kernel/projects/rt/4.6/older/patch-4.6-rc7-rt1.patch.gz
Applying the patch with
`zcat patch-4.6-rc7-rt1.patch.gz | patch -p1`
results in a lot of patching errors. Is this the right way and the kernel is modifyied such that it is not possible to apply the patch or did I made some mistakes?

Greetings from Ilmenau (Germany),
Fabian

umiddelb, 2016-06-13:
Hallo Fabian,

das ioft repository ist schon bei 4.7-rc3. Sobald 4.7 offiziell released wurde, branche ich noch einmal, so dass dann hoffentlich die rt patches dazu passen (werden). Alternativ könntest Du die original 4.6.2 Kernel Quellen nehmen und nur die xu4 defconfig aus dem ioft repository verwenden. Die einzige Einbuße ist der Umstand, dass das Ethernet Interface nur als USB 2.0 und nicht als USB 3.0 Gerät erkannt wird, der Rest sollte auf Anhieb gehen.

Viele Grüße aus Hamburg
Uli

jobenvil, 2016-06-13:
Hello Uli,

welche unterschiede besteht zwischen der Mainline und Master branche bei dem ioft Repository? Habe gerade master (4.6.0+) kompiliert. Wer betreut das? Fügst du die Patches hinein?

Schöne Grüße

umiddelb, 2016-06-14:
Hallo José,

ioft ist mein Zweitaccount, da pro Account nur ein Fork eines Basisrepositorys (z.B. torvalds/linux) möglich ist.

- [mainline](https://github.com/ioft/linux/tree/mainline) ist eine 1:1 Kopie von [torvalds/linux](https://github.com/torvalds/linux), d.h. ohne Patches

- [master](https://github.com/ioft/linux/tree/master) ist der eingefrorene Stand von 4.6, kombiniert mit Patches für XU4, CuBox-i und Utilite (Pro).

- [new_master](https://github.com/ioft/linux/tree/new_master) ist [mainline](https://github.com/ioft/linux/tree/mainline), kombiniert mit Patches für XU4, CuBox-i und Utilite (Pro).

Am liebsten würde ich nur [master](https://github.com/ioft/linux/tree/master) pflegen, doch von Zeit zu Zeit führen commits in [torvalds/linux](https://github.com/torvalds/linux) zu [Merge-Konflikten](https://github.com/ioft/linux/compare/master...torvalds:master).

Viele Grüße
Uli

jobenvil, 2016-06-15:
Moin Uli  `;-)`
vielen Dank für die erweiterte Erklärungen. 
> ioft ist mein Zweitaccount, 

habe ich mir auch gedacht, könnte aber der Name `ioft` nicht nachvollziehen.

Ich finde es Prima dass du Dir die Mühe machst und das so weiter pflegst. Genauso für die grossartige Anleitungen und weitere Erläuterungen in die HOWTOs.

Die Sache mit der Branches und Repositories ist hervorragend, aber am Ende verliert man den Überblickt, wer welche Branche pflegt und ob er alle Patches hat. Falls du Lust und Laune hast, könntest auch en bisschen über Patches erklären oder z.B. nach eine Änderung in der `.config`-File welche Schritte nötwendige sind um nicht weiter alles zu kompilieren mussen.

Schöne Grüße
José

umiddelb, 2016-06-22:
> Die Sache mit der Branches und Repositories ist hervorragend, aber am Ende verliert man den Überblickt, wer welche Branche pflegt und ob er alle Patches hat. Falls du Lust und Laune hast, könntest auch ein bisschen über Patches erklären oder z.B. nach eine Änderung in der .config-File welche Schritte notwendige sind um nicht weiter alles zu kompilieren müssen.

Die [Unterschiede zwischen den branches](https://github.com/torvalds/linux/compare/master...ioft:new_master) zeigt Dir GitHub ganz gut, so dass Du die einzelnen pull-requests nachvollziehen kannst. Für den XU4 ist es im wesentlichen USB3, für Utilite und CuBox-i der device tree. Und für alle drei die default config. Sobald 4.7 released ist wird der `new_master` zum `master` branch. Du kannst versuchen, nach Änderungen der `.config` das `make clean` wegzulassen. Ich bin mir aber nicht sicher, wie gut die make Abhängigkeiten gepflegt sind, insbesondere bei den BSP Kerneln.

jobenvil, 2016-06-22:
Uli, danke Dir nochmal. Ich bin mit so viele Änderungsmöglichkeiten im `exynos_defconfig` und `odoridxu4_defconfig` verwirrt. Jetzt habe ich wieder blaues Heartbeat-LED eingeschaltet und Lüfter lauf nicht standig an, bootet aber dafür nicht ;-) . habe gerade benutzt **@tombridan** master branch (4.7.rc-4). Ich brauche unbedingt einen UART-Serial-Kabel, bis jetzt bin ich mit VGA/HDMI-Konvertar klar gekommen, aber jetzt ist meinen Bildschirm Pechschwarz. Du kannst gerne das Issue schliessen. Teilt uns bitte deine OdroidXU4-Fortschritte im Forum mit.

umiddelb, 2016-06-22:
I don't care about spinning fans since I've attached a passive headsink to the XU4 ;) 
You definitely need serial access, but you can use any 1.7Volt UART USB converter in combination with three breadboard cables.

FabianBeck, 2016-07-08:
Hallo Uli,

tut mir leid, dass ich dir erst jetzt antworte aber ich bin nicht vorher dazu gekommen. 
Ich habe deinen Tipp befolgt und den originalen 4.6.2 Kernel mit der ioft config und rt patch gebaut. Soweit hat bisher auch alles geklappt und der Kernel hat mit dem minimalen Ubuntu 16.04 file system auch auf anhieb gebootet. Leider stürzt das Board zeitweise willkürlich ab, manchmal läuft es etliche Stunden durch und mal tritt es bereits nach 5 min auf. Eine Abschaltung wegen Überhitzung kann ich eigentlich ausschließen, da der Fehler auch ohne wirkliche Rechenlast auftritt und das Board nicht wirklich warm ist. Eine Auffälligkeit beim Boot in dmesg/seriellen Konsole war jedoch diese hier:

`[    0.003886] CPU: Testing write buffer coherency: ok
[    0.004699] CPU0: update cpu_capacity 448
[    0.005033] CPU0: thread -1, cpu 0, socket 1, mpidr 80000100
[    0.005253] Setting up static identity map for 0x40100000 - 0x40100058
[    0.005754] ARM CCI driver probed
[    0.006026] Exynos MCPM support installed
[    0.080368] CPU1: update cpu_capacity 448
[    0.080375] CPU1: thread -1, cpu 1, socket 1, mpidr 80000101
[    0.105344] CPU2: update cpu_capacity 448
[    0.105351] CPU2: thread -1, cpu 2, socket 1, mpidr 80000102
[    0.130344] CPU3: update cpu_capacity 448
[    0.130350] CPU3: thread -1, cpu 3, socket 1, mpidr 80000103
[    0.155401] CPU4: update cpu_capacity 1535
[    0.155407] CPU4: thread -1, cpu 0, socket 0, mpidr 80000000
[    0.180379] CPU5: update cpu_capacity 1535
[    0.180385] CPU5: thread -1, cpu 1, socket 0, mpidr 80000001
[    0.205382] CPU6: update cpu_capacity 1535
[    0.205387] CPU6: thread -1, cpu 2, socket 0, mpidr 80000002
[    0.230367] CPU7: update cpu_capacity 1535
[    0.230373] CPU7: thread -1, cpu 3, socket 0, mpidr 80000003
[    0.230650] Brought up 8 CPUs
[    0.231021] SMP: Total of 8 processors activated (384.00 BogoMIPS).
[    0.231055] CPU: WARNING: CPU(s) started in wrong/inconsistent modes (primary CPU mode 0x1a)
[    0.231100] CPU: This may indicate a broken bootloader or firmware.`

Jedoch scheint dies "normal" zu sein, da es auch bei Hardkernel im Mainline kernel (Kernel 4.2.0 RC1) auftritt wie hier im Log file zu sehen ist: 
http://www.hardkernel.com/main/products/prdt_info.php?g_code=G143452239825

Da ich den Kernel cross compiled habe, habe ich mich nicht exakt an deine Anleitung gehalten und bin mir nicht hundertprozentig sicher, dass bei der Einrichtung auf der SD Karte wirklich alles korrekt war. Weiß du etwas genaueres zu der obigen CPU Warning?

Da wir auf die neuen Features nicht zwingend angewiesen sind und wir doch Ubuntu 14.04 nutzen wollen werde ich als nächstes erstmal einen älteren Kernel mit rt-patch mit Ubuntu 14.04 file system ausprobieren und hoffen, dass dort der Fehler nicht auftritt.

viele Grüße,
Fabian

umiddelb, 2016-07-12:
Hallo Fabian,

> [ 0.231055] CPU: WARNING: CPU(s) started in wrong/inconsistent modes (primary CPU mode 0x1a)

Diese Warnung kenne ich auch, hat aber bei mir keinen negativen Impact zur Folge. Hast Du zufällig noch einen Dump der console von unmittelbar vor dem crash? Mit welcher gcc Version übersetzt Du den Kernel?

closed 2016-09-08 by umiddelb