umiddelb/armhf #26: Getting failure when running sudo sh build.sh

CarlosTorrecillas, 2016-08-17:
I'm getting the following error when running STEP 5 of the build.sh:

curl: (56) SSL read: error:00000000:lib(0):func(0):reason(0), errno 104

gzip: stdin: unexpected end of file
tar: Unexpected EOF in archive
tar: Unexpected EOF in archive
tar: Error is not recoverable: exiting now
The command '/bin/sh -c curl -sSL https://github.com/hypriot/golang-armbuilds/releases/download/v${GO_VERSION}/go${GO_VERSION}.linux-armv7.tar.gz | tar -xz -C /usr/local' returned a non-zero code: 2

I'm running it using on a Raspberry Pi Zero.

Any ideas?

Thanks,
Carlos.

umiddelb, 2016-08-17:
Hm, building Docker requires 2+ GB of RAM (including swap). 
May be the Docker build chain doesn't support ARMv6. 

Please raise your question [here](https://gitter.im/hypriot/talk).

closed 2016-08-17 by umiddelb

CarlosTorrecillas, 2016-08-22:
Looks like it is possible to run docker on a PiZero using HypriotOs:

https://twitter.com/CarlTorrecillas/status/766012465977196544

umiddelb, 2016-08-22:
> Looks like it is possible to run docker on a PiZero using HypriotOs:

That it is what it's made for ;), but it is very unlikely to **build** Docker 1.12 on the PiZero using HypriotOS.

CarlosTorrecillas, 2016-08-22:
Oh I see what you mean. Thanks for the info, it now makes sense!