umiddelb/armhf #27: unmaintained?

tve, 2016-09-02:
I'm looking for debian base images to build armhf container images and I've really liked your work. However, I notice that you have not updated this repo and the images on dockerhub in a while. Have you abandoned or paused this effort, and if so, which base images would you recommend to use instead? (I'm using odroids and am not interested in raspian base images.) Thanks much!!

umiddelb, 2016-09-02:
You're right. I'm not that enthusiastic any more providing very latest Debian base images ;) 
In the meantime Docker has assumed the task to provide Debian and Ubuntu base images for ARM
- https://hub.docker.com/r/armhf/debian/
- https://hub.docker.com/r/aarch64/debian/
 
Btw. you can still build you own one following the instructions mentioned [here](https://hub.docker.com/r/armv7/armhf-debian/).

tve, 2016-09-02:
Ok, thanks for the response and for having maintained this for a while :-)

closed 2016-09-02 by tve