umiddelb/armhf #29: Cannot start docker container

nephilim75, 2016-09-11:
Hello

I need some assistance on an issue that I am facing. I followed your instructions and to it looks like it is working. Please see this [record](http://paste.ubuntu.com/23163794/).

Now I try to run a container called Collabora. I used this command:
`docker run -t -d -p 9980:9980 -e "domain=sub\.domain\.com" --cap-add MKNOD collabora/code`

but it fails. Here is the [error log](http://paste.ubuntu.com/23163924/).

Any idea how to get this fixed? Or just a hint where I can start to investigate? Is this might be an collabora container issue?

Your assistance is highly appreciated.

Kind regards
//nephilim

kennethlimcp, 2016-09-11:
You will need to use `armhf` compatible images like https://hub.docker.com/r/container4armhf/armhf-alpine/

nephilim75, 2016-09-11:
Not sure, if I handled it the right way. Pls see the [error log](http://paste.ubuntu.com/23165589/) I received now. Did I miss anything?

umiddelb, 2016-09-11:
> Error response from daemon: Cannot start container 839c5c152137ff9acbe51bbf1a72e88f9717b1cdc21f5cf84d4472cd6208c017: [8] System error: exec format error

The binary in your image isn't made for arm (probably it's made for x86). You need to find out the Dockerfile for `collabora/code` and replace the base image by its armhf equivalent.

nephilim75, 2016-09-13:
Sorry to say, but I do not know how. I am newbie on this. Any hint for me?

umiddelb, 2016-09-13:
Hm, you may ask the creator of the Docker image to provide a version for armhf [here](https://www.collaboraoffice.com/code/) or [here](https://hub.docker.com/r/collabora/code/), since they don't disclose the build procedure. :(

nephilim75, 2016-09-14:
I will try this. Wrote them today. Will get back once I got a reply. Thank you.

closed 2016-09-25 by umiddelb