umiddelb/armhf #30: Update the XU4 example with the 4.14 branch

mad-ady, 2018-02-07:
Hi, could you update your XU4 example with the 4.14 branch from Hardkernel instead of 4.9?
It needs a branch change and also you can make odroidxu4_defconfig instead of xu3_defconfig.

Thanks!

umiddelb, 2018-02-07:
Hi Adrian,

the wiki page is [open to changes by everyone](https://github.com/umiddelb/armhf/wiki/How-To-compile-a-custom-Linux-kernel-for-your-ARM-device/_history). Just change it the way you think it should be, you don't need to ask for permission ;)  

Cheers
Uli

umiddelb, 2018-02-07:
OK, done.

Also removed the install_firmware target and added the openssl dev library. If I find the time I'll have to make an update to the whole wiki page, since Ubuntu 14.04 and gcc-4 have become very outdated.

closed 2018-02-07 by umiddelb

mad-ady, 2018-02-08:
Thanks. You're right - I could have edited it myself. Sorry :)