umiddelb/armhf #32: Embedding Linux headers while building custom android kernel bootable image

spaul13, 2018-07-31:
Just want to know, while building the custom Android kernel (arm64 architecture), is it possible to install corresponding Linux-headers with that.

As I want to install a few drivers software on Android and they are finding the Linux-headers corresponding to the android-kernel version? Can anyone please tell me how to do that?

umiddelb, 2018-08-02:
Hi,

I have no idea about the way how the Android Linux kernel is build. The 'vanilla' Linux source tree has a specific [make target](https://www.kernel.org/doc/makehelp.txt), `make headers_install ´ will install the header files at their default location. IMHO Android introduces their own defaults w.r.t. the file system layout, so the Linux defaults may not apply.

spaul13, 2018-08-02:
I have to build the custom Android kernel on Linux. Actually, the problem is while doing 'make headers install',  it installs the headers for the Linux not for my built custom Android kernel.

umiddelb, 2018-08-03:
I've no idea. You might better raise your question elsewhere.

closed 2018-08-03 by umiddelb