umiddelb/armhf #35: Could not resolve host

jalakoo, 2019-11-30:
I get the following error on a Raspberry Pi Zero running Raspbian Stretch in part 4 of the instructions from this [page](https://github.com/umiddelb/armhf/wiki/Get-Docker-up-and-running-on-the-RaspberryPi-%28ARMv6%29-in-three-steps):

`curl -sSL http://downloads.hypriot.com/docker-hypriot_1.8.2-1_armhf.deb >/tmp/docker-hypriot_1.8.2-1_armhf.deb
curl: (6) Could not resolve host: downloads.hypriot.com`

Thore-Krug, 2019-12-02:
1. Did u checked your Internet connection ? 2. Make sure the server u want to stuff from is still online !

umiddelb, 2019-12-02:
The hypriot people have moved the docker binaries to a different location in the meantime. You can find them [here](https://blog.hypriot.com/downloads/)(Hypriot Docker Debian Packages for Raspberry Pi). 

Newer Debian/Raspbian releases don't require a separate package download anymore since Docker has become an integral part of the operation system.

closed 2019-12-02 by umiddelb